lnagel/hass-komfovent #5: need help

latek0, 2025-03-06:
### System Health details

## System Information

version | core-2025.3.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.62-haos-raspi
arch | aarch64
timezone | Europe/Moscow
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 5000
Installed Version | 2.0.5
Stage | running
Available Repositories | 1575
Downloaded Repositories | 7

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.2
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 57.3 GB
disk_used | 8.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (7.0.0), File editor (5.8.0), Mosquitto broker (6.5.0), Zigbee2MQTT (2.1.3-1), Home Assistant Google Drive Backup (0.112.1), Terminal & SSH (9.16.0), TubesZB TI CC2652 FW Flasher (0.3.5.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 6
views | 5
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), end0 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth59c7faf (disabled), vethd33b146 (disabled), vethda942d9 (disabled), veth500f5cc (disabled), vethb5684d9 (disabled), vethadfdc71 (disabled), veth99011b6 (disabled), vethde4b4a4 (disabled), veth3940d18 (disabled), vethf4696ee (disabled), vethad49146 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), end0 (192.168.0.138/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth59c7faf (), vethd33b146 (), vethda942d9 (), veth500f5cc (), vethb5684d9 (), vethadfdc71 (), veth99011b6 (), vethde4b4a4 (), veth3940d18 (), vethf4696ee (), vethad49146 ()
ipv6_addresses | lo (::1/128), end0 (fd35:ca4a:b5df:2a81:92a8:4497:b5de:3556/64, fe80::795e:338c:f71a:4a/64), hassio (fe80::42:bdff:fe67:24ce/64), docker0 (fe80::42:2eff:fec1:1f46/64), veth59c7faf (fe80::641d:35ff:fef4:b288/64), vethd33b146 (fe80::9c62:1eff:fe35:d5eb/64), vethda942d9 (fe80::1007:efff:fe00:37ed/64), veth500f5cc (fe80::e483:f7ff:feb9:9fc0/64), vethb5684d9 (fe80::84bc:bbff:fe4b:db51/64), vethadfdc71 (fe80::a805:f7ff:fe02:1603/64), veth99011b6 (fe80::4c46:8bff:fe55:3195/64), vethde4b4a4 (fe80::6c5a:32ff:fe7b:b8f4/64), veth3940d18 (fe80::d49a:9bff:fe86:6100/64), vethf4696ee (fe80::f808:15ff:fec1:1ace/64), vethad49146 (fe80::208b:28ff:feab:5187/64)
announce_addresses | 192.168.0.138, fd35:ca4a:b5df:2a81:92a8:4497:b5de:3556, fe80::795e:338c:f71a:4a

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | February 19, 2025 at 22:39
-- | --
current_recorder_run | March 7, 2025 at 01:50
estimated_db_size | 201.60 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

error when add integration

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Unexpected error fetching komfovent data
Traceback (most recent call last):
  File "/config/custom_components/komfovent/modbus.py", line 50, in read_holding_registers
    raise ModbusException(f"Error reading registers at {address}")
pymodbus.exceptions.ModbusException: Modbus Error: Error reading registers at 999

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/komfovent/coordinator.py", line 101, in _async_update_data
    firmware_block = await self.client.read_holding_registers(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        registers.REG_FIRMWARE, 6
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/komfovent/modbus.py", line 57, in read_holding_registers
    raise ConfigEntryNotReady from error
homeassistant.exceptions.ConfigEntryNotReady: Modbus Error: Error reading registers at 999

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/komfovent/coordinator.py", line 110, in _async_update_data
    raise ConfigEntryNotReady from error
homeassistant.exceptions.ConfigEntryNotReady: Modbus Error: Error reading registers at 999
```

lnagel, 2025-03-07:
Please share the following to better understand the issue:
- the model of your device, controller type and firmware version
- a data dump taken with the `modbus_dump.py` tool
- screenshots of the data in the Komfovent app or web interface

lnagel, 2025-03-08:
I have implemented https://github.com/lnagel/hass-komfovent/pull/6 to help in obtaining this information. After you update the integration code, you should be able to click DOWNLOAD DIAGNOSTICS on the device page. This would help me to investigate.

krishels, 2025-03-08:
I have the same error
Modbus Error: Error reading registers at 999

krishels, 2025-03-08:
File "/config/custom_components/komfovent/modbus.py", line 57, in read_holding_registers
    raise ConfigEntryNotReady from error
homeassistant.exceptions.ConfigEntryNotReady: Modbus Error: Error reading registers at 999

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 380, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/komfovent/coordinator.py", line 110, in _async_update_data
    raise ConfigEntryNotReady from error
homeassistant.exceptions.ConfigEntryNotReady: Modbus Error: Error reading registers at 999
2025-03-08 22:39:13.119 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-03-08 22:39:13.220 ERROR (MainThread) [custom_components.komfovent.modbus] Failed to read registers at 0: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.107:502]
2025-03-08 22:39:13.221 ERROR (MainThread) [custom_components.komfovent.coordinator] Error communicating with Komfovent: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.107:502]
2025-03-08 22:39:13.221 ERROR (MainThread) [custom_components.komfovent.coordinator] Unexpected error fetching komfovent data
Traceback (most recent call last):
  File "/config/custom_components/komfovent/modbus.py", line 46, in read_holding_registers
    result = await self.client.read_holding_registers(
                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address, count=count, slave=1
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )

After this error, I have to disable the integration; otherwise, it crashes my recuperator web, and I cannot access it.

lnagel, 2025-03-08:
Could either of you help by sharing the diagnostic download file OR by running the modbus_dump.py tool and sharing its results?

krishels, 2025-03-11:
`{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2025.3.1",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
    "python_version": "3.13.2",
    "docker": true,
    "arch": "x86_64",
    "timezone": "Europe/Riga",
    "os_name": "Linux",
    "os_version": "6.6.73-haos",
    "supervisor": "2025.03.2",
    "host_os": "Home Assistant OS 14.2",
    "docker_version": "27.2.0",
    "chassis": "embedded",
    "run_as_root": true
  },
  "custom_components": {
    "dreame_vacuum": {
      "documentation": "https://github.com/Tasshack/dreame-vacuum",
      "version": "v2.0.0b16",
      "requirements": [
        "pillow",
        "numpy",
        "pybase64",
        "requests",
        "pycryptodome",
        "python-miio",
        "py-mini-racer",
        "paho-mqtt"
      ]
    },
    "pyscript": {
      "documentation": "https://github.com/custom-components/pyscript",
      "version": "1.6.1",
      "requirements": [
        "croniter==2.0.2",
        "watchdog==2.3.1"
      ]
    },
    "komfovent": {
      "documentation": "https://github.com/lnagel/hass-komfovent",
      "version": "0.1.0",
      "requirements": []
    },
    "volvo_cars": {
      "documentation": "https://github.com/thomasddn/ha-volvo-cars",
      "version": "1.4.3",
      "requirements": []
    },
    "mail_and_packages": {
      "documentation": "https://github.com/moralmunky/Home-Assistant-Mail-And-Packages",
      "version": "0.4.2",
      "requirements": [
        "beautifulsoup4",
        "Pillow>=9.0",
        "dateparser"
      ]
    },
    "life360": {
      "documentation": "https://github.com/pnbruckner/ha-life360/blob/0.5.4/README.md",
      "version": "0.5.4",
      "requirements": [
        "life360==7.0.1"
      ]
    },
    "aquarea": {
      "documentation": "https://github.com/cjaliaga/home-assistant-aquarea",
[... 1434 lines cut ...]
          "remaining": 10000,
          "reset": 1741708972
        },
        "scim": {
          "limit": 15000,
          "used": 0,
          "remaining": 15000,
          "reset": 1741708972
        },
        "dependency_snapshots": {
          "limit": 100,
          "used": 0,
          "remaining": 100,
          "reset": 1741705432
        }
      },
      "rate": {
        "limit": 5000,
        "used": 19,
        "remaining": 4981,
        "reset": 1741707523
      }
    }
  }
}`

krishels, 2025-03-11:
[config_entry-hacs-01J3G54SZKRMXDMY03QTMPAFAX (1).json](https://github.com/user-attachments/files/19186406/config_entry-hacs-01J3G54SZKRMXDMY03QTMPAFAX.1.json)

jiberan, 2025-03-11:
you can run like this: (change host to IP of your device)

` python3 modbus_dump.py --host 192.168.10.5`

krishels, 2025-03-11:
where i should run it? in HA terminal?

lnagel, 2025-03-12:
I have implemented a few improvements to the connect / data update methods that should make it run even if it cannot read the firmware versions. The changes from https://github.com/lnagel/hass-komfovent/pull/9 have just been merged.

Please re-download the integration in HACS to get the latest version, then after restarting, please share again the diagnostic data json file. I will be then able to understand better what is failing for you.

krishels, 2025-03-12:
This version works! 
Thx

krishels, 2025-03-12:
Could you please implement this section, where you can set tup temperature and flow levels?

[image]

lnagel, 2025-03-12:
> This version works! Thx

Please share with me the diagnostics JSON file so I can continue supporting your device type.

krishels, 2025-03-12:
> > This version works! Thx
> 
> Please share with me the diagnostics JSON file so I can continue supporting your device type.

[config_entry-hacs-01J3G54SZKRMXDMY03QTMPAFAX (2).json](https://github.com/user-attachments/files/19211037/config_entry-hacs-01J3G54SZKRMXDMY03QTMPAFAX.2.json)

latek0, 2025-03-12:
After updating, my version also starts to work.
What cards are you using in dashboard for vent control?
P.S. also attached my diagnostic file.

[config_entry-komfovent-01JP5ARVNPX60NH29P4T8BXC01.json](https://github.com/user-attachments/files/19211559/config_entry-komfovent-01JP5ARVNPX60NH29P4T8BXC01.json)

krishels, 2025-03-12:
Before this service, i used custom komfovent solution from HA forum. This is my dashboard yaml:

cards:
  - type: vertical-stack
    cards:
      - type: horizontal-stack
        cards:
          - type: custom:button-card
            entity: sensor.komfovent_operation_mode
            color: grey
            tap_action:
              action: call-service
              service: modbus.write_register
              service_data:
                address: 4
                hub: Komfovent
                unit: 1
                value: 1
            hold_action:
              action: none
            name: Away
            show_name: true
            icon: mdi:home-export-outline
            color_type: card
            state:
              - value: Away
                color: CornflowerBlue
          - type: custom:button-card
            entity: sensor.komfovent_operation_mode
            color: grey
            tap_action:
              action: call-service
              service: modbus.write_register
              service_data:
                address: 4
                hub: Komfovent
                unit: 1
                value: 2
            hold_action:
              action: none
            name: Normal
            show_name: true
            icon: mdi:home-account
            color_type: card
            state:
              - value: Normal
                color: CornflowerBlue
          - type: custom:button-card
            entity: sensor.komfovent_operation_mode
            color: grey
            tap_action:
              action: call-service
              service: modbus.write_register
              service_data:
                address: 4
                hub: Komfovent
                unit: 1
                value: 3
            hold_action:
              action: none
            name: Intensive
            show_name: true
            icon: mdi:weather-windy
            color_type: card
            state:
              - value: Intensive
                color: CornflowerBlue
          - type: custom:button-card
            entity: sensor.komfovent_operation_mode
            color: grey
            tap_action:
              action: call-service
              service: modbus.write_register
              service_data:
                address: 4
                hub: Komfovent
                unit: 1
                value: 4
            hold_action:
              action: none
            name: Boost
            show_name: true
            icon: mdi:fan
            color_type: card
            state:
              - value: Boost
                color: CornflowerBlue
      - type: horizontal-stack
        cards:
          - type: custom:button-card
            entity: sensor.komfovent_operation_mode
            color: grey
            tap_action:
              action: call-service
              service: shell_command.komfovent_kitchen_mode
              service_data: {}
              target: {}
            hold_action:
              action: none
            name: Kitchen
            show_name: true
            icon: mdi:atom-variant
            color_type: card
            state:
              - value: Kitchen
                color: CornflowerBlue
          - type: custom:button-card
            entity: switch.komfovent_switch_eco
            color: grey
            name: Eco
            icon: mdi:snowflake-variant
            hold_action: more-info
            color_type: card
            state:
              - value: "on"
                color: CornflowerBlue
              - value: "off"
                color: grey
          - type: custom:button-card
            entity: switch.komfovent_switch_auto
            color: grey
            hold_action:
              action: none
            name: Auto
            show_name: true
            icon: mdi:autorenew
            color_type: card
            state:
              - value: "on"
                color: CornflowerBlue
              - value: "off"
                color: grey
      - type: horizontal-stack
        cards:
          - type: sensor
            entity: sensor.komfovent_supply_temperature_c
            name: Supply
            detail: 1
          - type: sensor
            entity: sensor.komfovent_heater_power_w
            name: Heater
            graph: line
            icon: mdi:radiator
          - hours_to_show: 24
            graph: none
            type: sensor
            entity: sensor.komfovent_supply_fan_intensivity
            name: Supply
            icon: mdi:fan
            detail: 1
      - type: horizontal-stack
        cards:
          - type: sensor
            entity: sensor.komfovent_extract_temperature_c
            name: Extract
            icon: mdi:home-export-outline
      - type: horizontal-stack
        cards:
          - type: sensor
            entity: sensor.komfovent_status_normal_setpoint_c
            icon: mdi:thermometer
            name: Setpoint
          - type: sensor
            entity: sensor.komfovent_outdoor_temperature_c
            name: Outside
            icon: mdi:thermometer
          - type: sensor
            entity: sensor.komfovent_supply_flow_volume
            name: Supply
            icon: mdi:fan
  - type: entities
    entities:
      - entity: sensor.komfovent_heat_recovery_w
      - entity: sensor.komfovent_heating_recovery_day_kwh
      - entity: sensor.komfovent_power_consumption_w
      - entity: sensor.komfovent_ahu_consumption_day_kwh
    state_color: false
title: Gaiss
icon: mdi:air-filter
path: gaiss
theme: Catppuccin Frappe
type: masonry

lnagel, 2025-03-12:
Thanks for confirming that the issue is fixed, and for the config entries. I will be able to make test-cases out this now to guarantee it remains fixed.

I will close the issue now, and if you feel like discussing the dashboard, you can do it here: https://github.com/lnagel/hass-komfovent/issues/11

lnagel, 2025-03-12:
> > > This version works! Thx
> > 
> > 
> > Please share with me the diagnostics JSON file so I can continue supporting your device type.
> 
> [config_entry-hacs-01J3G54SZKRMXDMY03QTMPAFAX (2).json](https://github.com/user-attachments/files/19211037/config_entry-hacs-01J3G54SZKRMXDMY03QTMPAFAX.2.json)

Could you please share the diagnostics of the komfovent device? Currently you uploaded it for your entire HACS setup, which is not the right thing.

closed 2025-03-12 by lnagel

krishels, 2025-03-12:
I did not know, that to download Diagnostics, you have use browser instead of app.

[config_entry-komfovent-01JP2VDDEHPV8VZK5CKH4EMQGE.json](https://github.com/user-attachments/files/19212708/config_entry-komfovent-01JP2VDDEHPV8VZK5CKH4EMQGE.json)

lnagel, 2025-03-12:
> I did not know, that to download Diagnostics, you have use browser instead of app.
> 
> [config_entry-komfovent-01JP2VDDEHPV8VZK5CKH4EMQGE.json](https://github.com/user-attachments/files/19212708/config_entry-komfovent-01JP2VDDEHPV8VZK5CKH4EMQGE.json)

Thanks, this file is good. For me download diagnostics link is also usable via mobile app:

[image]

vahesoo, 2025-03-16:
Hi, havnt had lot of time to test everything out yet, but noticed strange behaviour. When outside temperature drops below 0 degrees then entity shows +6553⁰c

[config_entry-komfovent-01JPCDX6W15P1WPRVXEXACFS5N.json](https://github.com/user-attachments/files/19274886/config_entry-komfovent-01JPCDX6W15P1WPRVXEXACFS5N.json)

[image]

lnagel, 2025-03-16:
> Hi, havnt had lot of time to test everything out yet, but noticed strange behaviour. When outside temperature drops below 0 degrees then entity shows +6553⁰c

That is fixed in the latest release, please upgrade.