lnagel/hass-komfovent #11: Dashboard discussion

lnagel, 2025-03-12:
> Before this service, i used custom komfovent solution from HA forum. This is my dashboard yaml:
> 
> 
> cards:
>   - type: vertical-stack
>     cards:
>       - type: horizontal-stack
>         cards:
>           - type: custom:button-card
>             entity: sensor.komfovent_operation_mode
>             color: grey
>             tap_action:
>               action: call-service
>               service: modbus.write_register
>               service_data:
>                 address: 4
>                 hub: Komfovent
>                 unit: 1
>                 value: 1
>             hold_action:
>               action: none
>             name: Away
>             show_name: true
>             icon: mdi:home-export-outline
>             color_type: card
>             state:
>               - value: Away
>                 color: CornflowerBlue
>           - type: custom:button-card
>             entity: sensor.komfovent_operation_mode
>             color: grey
>             tap_action:
>               action: call-service
>               service: modbus.write_register
>               service_data:
>                 address: 4
>                 hub: Komfovent
>                 unit: 1
>                 value: 2
>             hold_action:
>               action: none
>             name: Normal
>             show_name: true
>             icon: mdi:home-account
>             color_type: card
>             state:
>               - value: Normal
>                 color: CornflowerBlue
>           - type: custom:button-card
>             entity: sensor.komfovent_operation_mode
>             color: grey
>             tap_action:
>               action: call-service
>               service: modbus.write_register
>               service_data:
>                 address: 4
>                 hub: Komfovent
>                 unit: 1
>                 value: 3
>             hold_action:
>               action: none
>             name: Intensive
>             show_name: true
>             icon: mdi:weather-windy
>             color_type: card
>             state:
>               - value: Intensive
>                 color: CornflowerBlue
>           - type: custom:button-card
>             entity: sensor.komfovent_operation_mode
>             color: grey
>             tap_action:
>               action: call-service
>               service: modbus.write_register
>               service_data:
>                 address: 4
>                 hub: Komfovent
>                 unit: 1
>                 value: 4
>             hold_action:
>               action: none
>             name: Boost
>             show_name: true
>             icon: mdi:fan
>             color_type: card
>             state:
>               - value: Boost
>                 color: CornflowerBlue
>       - type: horizontal-stack
>         cards:
>           - type: custom:button-card
>             entity: sensor.komfovent_operation_mode
>             color: grey
>             tap_action:
>               action: call-service
>               service: shell_command.komfovent_kitchen_mode
>               service_data: {}
>               target: {}
>             hold_action:
>               action: none
>             name: Kitchen
>             show_name: true
>             icon: mdi:atom-variant
>             color_type: card
>             state:
>               - value: Kitchen
>                 color: CornflowerBlue
>           - type: custom:button-card
>             entity: switch.komfovent_switch_eco
>             color: grey
>             name: Eco
>             icon: mdi:snowflake-variant
>             hold_action: more-info
>             color_type: card
>             state:
>               - value: "on"
>                 color: CornflowerBlue
>               - value: "off"
>                 color: grey
>           - type: custom:button-card
>             entity: switch.komfovent_switch_auto
>             color: grey
>             hold_action:
>               action: none
>             name: Auto
>             show_name: true
>             icon: mdi:autorenew
>             color_type: card
>             state:
>               - value: "on"
>                 color: CornflowerBlue
>               - value: "off"
>                 color: grey
>       - type: horizontal-stack
>         cards:
>           - type: sensor
>             entity: sensor.komfovent_supply_temperature_c
>             name: Supply
>             detail: 1
>           - type: sensor
>             entity: sensor.komfovent_heater_power_w
>             name: Heater
>             graph: line
>             icon: mdi:radiator
>           - hours_to_show: 24
>             graph: none
>             type: sensor
>             entity: sensor.komfovent_supply_fan_intensivity
>             name: Supply
>             icon: mdi:fan
>             detail: 1
>       - type: horizontal-stack
>         cards:
>           - type: sensor
>             entity: sensor.komfovent_extract_temperature_c
>             name: Extract
>             icon: mdi:home-export-outline
>       - type: horizontal-stack
>         cards:
>           - type: sensor
>             entity: sensor.komfovent_status_normal_setpoint_c
>             icon: mdi:thermometer
>             name: Setpoint
>           - type: sensor
>             entity: sensor.komfovent_outdoor_temperature_c
>             name: Outside
>             icon: mdi:thermometer
>           - type: sensor
>             entity: sensor.komfovent_supply_flow_volume
>             name: Supply
>             icon: mdi:fan
>   - type: entities
>     entities:
>       - entity: sensor.komfovent_heat_recovery_w
>       - entity: sensor.komfovent_heating_recovery_day_kwh
>       - entity: sensor.komfovent_power_consumption_w
>       - entity: sensor.komfovent_ahu_consumption_day_kwh
>     state_color: false
> title: Gaiss
> icon: mdi:air-filter
> path: gaiss
> theme: Catppuccin Frappe
> type: masonry
>  

 _Originally posted by @krishels in [#5](https://github.com/lnagel/hass-komfovent/issues/5#issuecomment-2718065877)_

latek0, 2025-03-12:
My buttons doesnt work in this card

[image]

latek0, 2025-03-12:
@lnagel 
Please advice how to use your dashboard in HA?

[image]

lnagel, 2025-03-12:
I have removed the dashboard from the source, it was not implemented.

purplesticks, 2025-03-14:
Hi, is it possible to rename the English names in **

[image]

** to others somewhere? I would like to rename the drop-down list to my language. If possible, please give me instructions. I am a beginner in HA. Thank you in advance for the add-on. It works perfectly :)

lnagel, 2025-03-30:
For the dashboard, please see if the included Climate entity card would work for you. It handles mode and mode-specific temperature setpoint control both. For me this seems much simpler and cleaner than a grid of mode buttons and setpoints.

Please see work in progress here: 
https://github.com/lnagel/hass-komfovent/pull/58

MaBeniu, 2025-09-14:
[image]

Sharing mine if some ideas needed:
```
type: vertical-stack
cards:
  - type: horizontal-stack
    cards:
      - show_name: true
        show_icon: true
        type: button
        entity: switch.komfovent_auto_mode
        show_state: true
        icon: mdi:fan-auto
        name: Auto
        icon_height: 30px
      - show_name: true
        show_icon: true
        type: button
        entity: switch.komfovent_eco_mode
        name: Eco
        show_state: true
        icon_height: 30px
      - type: custom:bubble-card
        card_type: button
        button_type: slider
        entity: sensor.komfovent_filter_clogging
        show_icon: false
        show_name: true
        force_icon: false
        show_state: true
        sub_button: []
        card_layout: large
        show_attribute: false
        show_last_updated: false
        rows: "1.5"
        name: Filter clogging
  - type: horizontal-stack
    cards:
      - graph: none
        type: sensor
        entity: sensor.temp1_temperature
        icon: mdi:thermometer
        detail: 1
        name: 2nd floor
      - graph: none
        type: sensor
        entity: sensor.daikin_outdoor_temp
        name: Outdoor
        icon: ""
        detail: 1
      - graph: none
        type: sensor
        entity: sensor.garagetemp_temperature
        icon: ""
        detail: 1
        name: Indoor
  - type: horizontal-stack
    cards:
      - type: custom:bubble-card
        card_type: button
        button_type: state
        entity: sensor.komfovent_supply_temperature
        show_icon: true
        show_name: true
        force_icon: false
        show_state: true
        sub_button:
          - state_background: false
            show_background: false
            show_name: false
            name: ""
            show_last_changed: false
            show_state: true
            show_attribute: false
            attribute: unit_of_measurement
            entity: sensor.komfovent_supply_fan
            icon: mdi:fan
        card_layout: large
        show_attribute: false
        show_last_updated: false
        rows: "1.5"
        name: Supply
        scrolling_effect: false
      - type: custom:bubble-card
        card_type: button
        button_type: state
        show_icon: true
        show_name: true
        force_icon: false
        show_state: true
        sub_button:
          - state_background: false
            show_background: false
            show_name: true
            name: ""
            show_last_changed: false
            show_state: true
            show_attribute: false
            attribute: unit_of_measurement
            entity: sensor.komfovent_extract_fan
            icon: mdi:fan
        card_layout: large
        show_attribute: false
        show_last_updated: false
        rows: "1.5"
        name: Extract
        scrolling_effect: false
        entity: sensor.komfovent_extract_temperature
  - type: horizontal-stack
    cards:
      - graph: line
        type: sensor
        entity: sensor.komfovent_heater_power
        name: Heater
        icon: mdi:radiator
        detail: 2
        hours_to_show: 24

```