lnagel/hass-komfovent #14: Translations for select entities

lnagel, 2025-03-14:
> Hi, is it possible to rename the english names in **Komfovent Current mode** to others somewhere? I would like to rename the drop-down list to my language. If possible, please give me instructions. I am a beginner in HA. Thank you in advance for the add-on. It works perfectly :)
> 
> [image]
>  

 _Originally posted by @purplesticks in [#11](https://github.com/lnagel/hass-komfovent/issues/11#issuecomment-2724621184)_

CodekExplor, 2025-03-19:
I would like help with the translation and learn something along the way. Is it possible to create a language file? I try to review your changes and am amazed at your work. Respect!

lnagel, 2025-03-29:
> I would like help with the translation and learn something along the way. Is it possible to create a language file? I try to review your changes and am amazed at your work. Respect!

Thanks for your kind words!

I am sure it is possible to create a translation file, but I have not figured out what exactly is the structure within the file to make this work correctly. I didn't find a specific example for this in any docs. Maybe scanning other custom component repositories might reveal how it's done.

ArekKubacki, 2025-05-10:
My version with translations :)

[komfovent.zip](https://github.com/user-attachments/files/20141115/komfovent.zip)

lnagel, 2025-05-13:
I am currently not yet ready to accept translation contributions, as the translation keys and texts needs to be prepared for translation. Before this is done, then I don't really have a way to integrate it.

ArekKubacki, 2025-05-13:
The package I sent is not a translation file. The Polish language is just an addition there and can be left as a note. The most important thing is that it fix the bug for select from 1 post in this thread.

lnagel, 2025-05-13:
Okay, as you wish. Just be prepared that there will be changes in the language strings once the system is in place.