lnagel/hass-komfovent #15: Exhaust & Flow rate

Gajer, 2025-03-15:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Hi,

First of all, thank you for this add-on! I really needed it!

I'm opening a ticket to request the addition of the exhaust air temperature from the heat recovery unit and the airflow rate. 

### Describe the solution you'd like

My programming knowledge is at a basic level, but if you have time, please point me in the right direction, and I’d be happy to help—I just don’t know where to start.

### Describe alternatives you've considered

None

### Additional context

None

lnagel, 2025-03-15:
@CodekExplor To confirm the availability of exhaust air temperature from your units, please take a diagnostic data file from this integration. If your device actually supports this, it could be added. I will note, however, that my device does not support it yet (maybe I have old firmware).

Please update the integration to the latest version and after that post the diagnostic json file here. The latest version will attempt to include this in the diagnostics.

CodekExplor, 2025-03-15:
Thank you for your answer, I'm uploading the test. I would like to note that the exhaust temperature value is unknown and I know that it works when I used the modbus module. And additionally, the outside temperature values ​​do not correctly indicate negative values. 

[config_entry-komfovent-01JPE1AZCW0D8GYA4SNVZBP1CD.json](https://github.com/user-attachments/files/19267072/config_entry-komfovent-01JPE1AZCW0D8GYA4SNVZBP1CD.json)

[image]

[image]

lnagel, 2025-03-16:
@CodekExplor Please could you share a screenshot of the firmware version in your Komfovent settings web view as well?

closed 2025-03-16 by lnagel

CodekExplor, 2025-03-16:
[image]

lnagel, 2025-03-17:
Thanks, I was able to add controller type detection based on this data:

Extract Controller type from first 4 bits of version #26