lnagel/hass-komfovent #31: Failed dependencies modbus

korzen123, 2025-03-17:
### System Health details

Core
2025.1.3
Supervisor
2025.03.3
Operating System
14.2
Interfejs użytkownika
20250109.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

"Failed dependencies modbus" error during starting up integration

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Rejestrator: homeassistant.setup
Źródło: setup.py:334
Pierwsze zdarzenie: 22:37:19 (1 zdarzenia)
Ostatnio zalogowany: 22:37:19

Setup failed for 'modbus': Unable to import component: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/modbus/__init__.py", line 142, in <module>
    from .modbus import ModbusHub, async_modbus_setup
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 17, in <module>
    from pymodbus.framer import FramerType
ImportError: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 334, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1035, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1027, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1075, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 200, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/modbus/__init__.py", line 142, in <module>
    from .modbus import ModbusHub, async_modbus_setup
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 17, in <module>
    from pymodbus.framer import FramerType
ImportError: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)
```

lnagel, 2025-03-18:
It seems something is wrong with your HA install. The core version seems out of date and the pymodbus library you have installed is incompatible with it.

In any case, I currently don't see how this is related to this integration component, as there is no mention of komfovent within the stack trace.

korzen123, 2025-03-18:
I did it manual, but now I upgraded core to 2025.3.3 and instaled HACS and I have the same problem "Failed dependencies modbus"

Rejestrator: homeassistant.setup
Źródło: setup.py:340
Pierwsze zdarzenie: 21:12:41 (1 zdarzenia)
Ostatnio zalogowany: 21:12:41

Setup failed for 'modbus': Unable to import component: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1014, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/modbus/__init__.py", line 146, in <module>
    from .modbus import DATA_MODBUS_HUBS, ModbusHub, async_modbus_setup
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 17, in <module>
    from pymodbus.framer import FramerType
ImportError: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 340, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1034, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1026, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1074, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/modbus/__init__.py", line 146, in <module>
    from .modbus import DATA_MODBUS_HUBS, ModbusHub, async_modbus_setup
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 17, in <module>
    from pymodbus.framer import FramerType
ImportError: cannot import name 'FramerType' from 'pymodbus.framer' (/usr/local/lib/python3.13/site-packages/pymodbus/framer/__init__.py)

lnagel, 2025-03-19:
It does seem like the python dependencies (specifically the pymodbus version) in your home assistant install are out of date and incompatible with the version of home assistant you are running. This indicates a deeper problem with your HA installation, which I am unable to assist with in the scope of this custom component. You'll need to fix this before attempting to run the Komfovent integration.

You might get some context from here: https://github.com/pymodbus-dev/pymodbus/discussions/2192

For example, in my install, the correct version of pymodbus is installed:

```
# pip freeze | grep modbus
async_modbus==0.2.2
pymodbus==3.8.3
```

closed 2025-03-19 by lnagel