lnagel/hass-komfovent #40: Provide the unit time as a sensor

borpin, 2025-03-23:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

Could you please supply the Unit time as a sensor.

### Describe the solution you'd like

Could you please supply the Unit time as a sensor.

### Describe alternatives you've considered

Nil

### Additional context

Nil

borpin, 2025-03-24:
Linked to #37

lnagel, 2025-03-29:
Support for this was released in v0.3.0