lnagel/hass-komfovent #41: Select System Mode via a button

borpin, 2025-03-23:
### Checklist

- [x] I have filled out the template to the best of my ability.
- [x] This only contains 1 feature request (if you have multiple feature requests, open one feature request for each feature request).
- [x] This issue is not a duplicate feature request of [previous feature requests](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Feature+Request%22+).

### Is your feature request related to a problem? Please describe.

On my existing system, I have buttons for the mode selection, like the Komfovent Web interface. It would be good to be able to replicate this.

```
type: custom:button-card
entity: sensor.komfovent_operation_mode
color: grey
tap_action:
  action: call-service
  service: modbus.write_register
  service_data:
    address: 4
    hub: Komfovent
    unit: 1
    value: 2
hold_action:
  action: none
name: Normal
show_name: true
icon: mdi:home-account
color_type: card
state:
  - value: Normal
    color: CornflowerBlue
```

[image]

### Describe the solution you'd like

Replicate the button interface on the Komfovent Web Interface

### Describe alternatives you've considered

Write to the Modbus directly

### Additional context

Nil

lnagel, 2025-03-26:
Theoretically, could you implement the buttons using the existing `komfovent_current_mode` select entity, simply by writing a new value to it? The constants `away`, `normal`, `intensive` and `boost` should be accepted.

lnagel, 2025-03-28:
Current draft for the `set_mode` action + button config, what do you think?

```
show_name: true
show_icon: true
type: button
entity: select.komfovent_current_mode
name: Normal
icon: mdi:home
show_state: false
tap_action:
  action: perform-action
  perform_action: komfovent.set_mode
  target: {}
  data:
    config_entry: 01JPCJ33S8XJWA4E31B8N6NKV9
    mode: normal
```

[image]

closed 2025-03-29 by lnagel

lnagel, 2025-03-29:
Support for this was released in v0.4.0 - see the README for examples

borpin, 2025-03-30:
I haven't checked, but does the README list the possible values for the data for the service?

lnagel, 2025-03-30:
It does