lnagel/hass-komfovent #43: Heat Recovery control showing AUTO when should be Constant

borpin, 2025-03-23:
### System Health details

Can't find the system health page. The documentation is not complete enough :(

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

There is an issue with the Heat Recovery control. It is showing AUTO when the register (217) is set to 1 (constant)

[image]

[image]

### Reproduction steps

1. View what the integration thinks the setting is.

### Debug logs

```text
I do see occasional errors when reading the address 216.

2025-03-23 10:05:47.105 WARNING (MainThread) [custom_components.komfovent.coordinator] Error communicating with Komfovent: Modbus Error: Error reading registers at 900
2025-03-23 10:06:13.022 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:06:13.023 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Komfovent: Error: device: 1 address: 216 -> pymodbus returned isError True
2025-03-23 10:06:17.095 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:06:17.096 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Komfovent: Error: device: 1 address: 0 -> pymodbus returned isError True
2025-03-23 10:06:17.110 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:06:17.111 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Komfovent: Error: device: 1 address: 3 -> pymodbus returned isError True
2025-03-23 10:06:17.116 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:06:47.099 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:06:47.099 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Komfovent: Error: device: 1 address: 0 -> pymodbus returned isError True
2025-03-23 10:06:47.102 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:06:47.110 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:17.104 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:17.104 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Komfovent: Error: device: 1 address: 0 -> pymodbus returned isError True
2025-03-23 10:07:17.107 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:17.114 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:17.117 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:47.108 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:47.108 ERROR (MainThread) [homeassistant.components.modbus.modbus] Pymodbus: Komfovent: Error: device: 1 address: 0 -> pymodbus returned isError True
2025-03-23 10:07:47.111 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:47.119 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:47.124 ERROR (MainThread) [pymodbus.logging] Exception response 131 / 0
2025-03-23 10:07:47.125 WARNING (MainThread) [custom_components.komfovent.coordinator] Error communicating with Komfovent: Modbus Error: Error reading registers at 900
2025-03-23 10:07:47.125 ERROR (MainThread) [custom_components.komfovent.coordinator] Error fetching komfovent data:
```

### Diagnostics dump

[config_entry-komfovent-01JPT0DH2A18SFCXCDTQNFR61N.json](https://github.com/user-attachments/files/19409329/config_entry-komfovent-01JPT0DH2A18SFCXCDTQNFR61N.json)

lnagel, 2025-03-25:
@borpin Do you by any chance have multiple integrations trying to communicate with the Komfovent AHU concurrently?

borpin, 2025-03-25:
> [@borpin](https://github.com/borpin) Do you by any chance have multiple integrations trying to communicate with the Komfovent AHU concurrently? The unit does not always respond well in this case. You might need to setup a modbus buffering proxy.

Yes I do (partly because of me testing this integration) and I do see occasional errors coming back because of it. What is a 'Modbus Buffering Proxy'?

lnagel, 2025-03-25:
Try this: https://github.com/tiagocoutinho/modbus-proxy

> Many modbus devices support only one or very few clients. This proxy acts as a bridge between the client and the modbus device. It can be seen as a layer 7 reverse proxy. This allows multiple clients to communicate with the same modbus device.
> 
> When multiple clients are connected, cross messages are avoided by serializing communication on a first come first served REQ/REP basis.

lnagel, 2025-03-25:
When I am developing this integration, I am mostly running fake Komfovent AHU modbus server in my laptop, just to preview how things will look. Then it's quite safe to try things out.

You could use the same approach to clone your real Komfovent device for testing and just run my integration against the clone.

```bash
modbus_dump.py --host=192.168.1.60 --port=502 --output=C6M_registers.json
modbus_server.py --host=0.0.0.0 --port=10502 --input=C6M_registers.json
```

lnagel, 2025-03-30:
I will close this as the underlying issue has been fixed. I have some plans for the future though to improve the handling of failed reads, though. It's a bit less than perfect to see the occasional Unknown value in the logs (maybe once per day for me).

closed 2025-03-30 by lnagel