lnagel/hass-komfovent #61: Error setting up entry Komfovent for komfovent

korzen123, 2025-03-31:
### System Health details

Core
2025.3.3
Supervisor
2025.03.4
Operating System
14.2
Interfejs użytkownika
20250306.0

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/lnagel/hass-komfovent/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

ERROR ON STARTUP, BELOW IS THE LOG FROM HA
WHAT DOES THIS MEAN

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
Rejestrator: homeassistant.config_entries
Źródło: config_entries.py:753
Pierwsze zdarzenie: 23:26:00 (1 zdarzenia)
Ostatnio zalogowany: 23:26:00

Error setting up entry Komfovent for komfovent
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 753, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/komfovent/__init__.py", line 120, in async_setup_entry
    coordinator = KomfoventCoordinator(
        hass, entry.data[CONF_HOST], entry.data[CONF_PORT]
    )
  File "/config/custom_components/komfovent/coordinator.py", line 35, in __init__
    from .modbus import KomfoventModbusClient  # easier to mock
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/komfovent/modbus.py", line 6, in <module>
    from pymodbus import ModbusException
ImportError: cannot import name 'ModbusException' from 'pymodbus' (/usr/local/lib/python3.13/site-packages/pymodbus/__init__.py)
```

lnagel, 2025-04-03:
I will proceed to close the issue as there has not been any feedback and currently there is no reason to believe the integration would be at fault. If you have further info that would suggest otherwise, please attach the diagnostic file and re-open the ticket. I'd be happy to look further into this.

closed 2025-04-03 by lnagel

korzen123, 2025-04-04:
After update HA to 
Core2025.3.4
Supervisor2025.03.4
Operating System15.1 

Working well:)
thx