lnagel/hass-komfovent #64: Failed to connect to Komfovent device

EvaldasD21, 2025-04-05:
### System Health details

## System Information

version | core-2025.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 6.6.74-haos-raspi
arch | aarch64
timezone | Europe/Vilnius
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4976
Installed Version | 2.0.5
Stage | running
Available Repositories | 1627
Downloaded Repositories | 32

</details>

<details><summary>AccuWeather</summary>

can_reach_server | ok
-- | --
remaining_requests | 37

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 15.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.03.4
agent_version | 1.7.2
docker_version | 28.0.4
disk_total | 116.6 GB
disk_used | 8.5 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | rpi5-64
supervisor_api | ok
[... 285 lines cut ...]
2025-04-05 22:18:21.692 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:18:40.597 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:19:25.599 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:19:26.694 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:20:31.697 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:20:33.456 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:20:48.458 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:21:02.846 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:21:13.459 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:21:17.848 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:21:42.850 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:21:58.460 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:22:27.853 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:23:03.463 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:23:35.591 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:23:50.593 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:24:08.467 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:24:15.594 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:25:00.595 WARNING (MainThread) [pymodbus.logging] Failed to connect 
2025-04-05 22:25:13.469 WARNING (MainThread) [pymodbus.logging] Failed to connect
```

### Diagnostics dump

[home-assistant_komfovent_2025-04-05T18-40-05.131Z.log](https://github.com/user-attachments/files/19617968/home-assistant_komfovent_2025-04-05T18-40-05.131Z.log)

lnagel, 2025-04-06:
@EvaldasD21 
Could you please copy the Controller firmware version number from your Komfovent web settings view? What type of device and controller do you have?

We will get this to work, but further info is required to debug the failed read command.

lnagel, 2025-04-06:
Please also upgrade the integration via HACS to [v0.5.1](https://github.com/lnagel/hass-komfovent/releases/tag/v0.5.1) and after that download the diagnostics. This should take a full snapshot with register data that is required for debugging this. Please attach that diagnostics json file here to the issue.

[image]

lnagel, 2025-04-06:
Do you have the possibility to run the modbus tools python scripts to take a snapshot of the device? This registers.json file would help me to support your device. You would need to download this code repo and run the script.

https://github.com/lnagel/hass-komfovent/blob/main/modbus_dump.py

The `modbus_dump.py` tool can be used to dump the ModBus data from the Komfovent device. Usage:

```bash
python3 modbus_dump.py --host <device_ip> [--port 502] [--output registers.json]
```

This will scan all known register ranges and save the results to a JSON file. The tool supports these arguments:
- `--host`: Required. IP address of your Komfovent device
- `--port`: Optional. ModBus TCP port (default: 502)
- `--output`: Optional. Output JSON file path (default: registers.json)

EvaldasD21, 2025-04-06:
> [@EvaldasD21](https://github.com/EvaldasD21)
> 
> We will get this to work, but further info is required to debug the failed read command.
> 
> * Could you please copy the Controller firmware version number from your Komfovent web settings view?
> * What type of device and controller do you have?
> 
> If you have not updated the firmware, then you might want to do this before continuing. Depending on the controller type, please see the correct instructions.
> 
> * [C6/C6M firmware update instructions](https://www.komfovent.com/en/downloads/C6_update_EN.pdf)
> * [C8 firmware update instructions](https://www.komfovent.lt/lt/references/C8_CONTROLLER_Firmware_update_EN.pdf)

Hi lnagel, 
thanks for your quick reply.
Thanks to your suggestions, I managed to connect the device and integration is working perfectly.
I did two things, not sure which one helped:

1. Although before I was able to access the device via browser, connection seemed to be unstable, often received browser error "page could not be reached", after few refreshes it finally loaded (slowly). So I tried to do direct connection to PC via ethernet port, that somehow did not work at all so I repluged local network cable back to the controller, and after this connection via browser was stable. 
2. After step 1 and stable connection via local network I have updated C8 controller firmware from C8 1.1.3.13 to the latest one C8 1.1.14.27.

After these two steps I have checked the integration on Home Assistant and it found the device with no problem. 
Thanks again for your ideas and a wonderful integration!

Evaldas

lnagel, 2025-04-06:
@EvaldasD21 Thanks for the response, this is really good news.

Since you are the first here with a C8 type device, could you assist me in supporting it by providing helpful diagnostics data? If you can send this, then I can make all the feature detections and inputs work correctly.

I would need the diagnostics file:

[image]

And also screenshots of all the pages in device webapp:

- Overview page
- Operation modes -> modify
- Auto/Eco modes -> modify
- Overview -> detailed information
- Settings

lnagel, 2025-04-06:
Click "Update integration" on the context menu settings in HACS.

[image]

Then an update will be offered in your main settings dashboard page.

EvaldasD21, 2025-04-07:
> [@EvaldasD21](https://github.com/EvaldasD21) Thanks for the response, this is really good news.
> 
> Since you are the first here with a C8 type device, could you assist me in supporting it by providing helpful diagnostics data? If you can send this, then I can make all the feature detections and inputs work correctly.
> 
> Before continuing, please upgrade via HACS to [v0.5.1](https://github.com/lnagel/hass-komfovent/releases/tag/v0.5.1)
> 
> I would need the diagnostics file:
> 
> [image]
> 
> And also screenshots of all the pages in device webapp:
> 
> * Overview page
> * Operation modes -> modify
> * Auto/Eco modes -> modify
> * Overview -> detailed information
> * Settings

Updated to v0.5.1
Here is diagnostics file.
[config_entry-komfovent-01JR3M6TGYCDF07DE3S60K01AJ.json](https://github.com/user-attachments/files/19627697/config_entry-komfovent-01JR3M6TGYCDF07DE3S60K01AJ.json)

If I understand correctly, you need screenshots of device's Web interface. If so, I will send you when I will be back at my local network.

lnagel, 2025-04-07:
Thanks! Please if possible, take another diagnostics file at the same time as the screenshots - then it's possible to confirm that the measurements match between the Komfovent UI and the data registers.

EvaldasD21, 2025-04-07:
> Thanks! Please if possible, take another diagnostics file at the same time as the screenshots - then it's possible to confirm that the measurements match between the Komfovent UI and the data registers. Please also put the unit in some regular operation mode, e.g. Normal or Intensive. Currently in there is no airflow data (unit in Standby).

Hi,
Sharing diagnostics file and screenshots as per your request.
P.S. After I tried to test controls in HA today, I noticed that ECO mode toggles but does nothing for the device. Also all entities of modes seem to be unavailable. Not sure about other settings if they are functional.
[image]

[config_entry-komfovent-01JR3M6TGYCDF07DE3S60K01AJ (1).json](https://github.com/user-attachments/files/19637587/config_entry-komfovent-01JR3M6TGYCDF07DE3S60K01AJ.1.json)

[image]

[image]

[image]

[image]

[image]

lnagel, 2025-04-08:
> Hi, Sharing diagnostics file and screenshots as per your request

Thanks, much appreciated. I will review these and adapt the integration to support C8 controllers.
Further details will be here in: **Add support for C8 controller type #66** 

> P.S. After I tried to test controls in HA today, I noticed that ECO mode toggles but does nothing for the device.

From your screenshot, I do see this actually taking effect. Since you have ECO max supply air temperature set to 20 C and the current supply temp is 22 C, then the controller has already entered the "Flow reduction" mode (visible also as the 3rd icon on the top left in the first overview page). You can see that your normal mode sets flow at 60%, but the controller has already reduced to 55% and possibly will fully stop at this point. This is what the controller does, that's all.

> Also all entities of modes seem to be unavailable. Not sure about other settings if they are functional. [image]

These entities are not provided by this integration. Please open Home Assistant Settings -> Devices & services -> Entities and filter by the Komfovent integration. Then you will see the entities provided.

EvaldasD21, 2025-04-08:
> From your screenshot, I do see this actually taking effect. Since you have ECO max supply air temperature set to 20 C and the current supply temp is 22 C, then the controller has already entered the [...]

What I meant by does nothing is that ECO mode can be toggled in HA, but that does not change ECO status on the device (ECO button highlighted when On and gray when Off), that I can see in WebApp, and vice versa, I can change ECO mode status on WebApp, but status in HA does not change.

lnagel, 2025-04-08:
> > > P.S. After I tried to test controls in HA today, I noticed that ECO mode toggles but does nothing for the device.
> > 
> > 
> > From your screenshot, I do see this actually taking effect. Since you have ECO max supply air temperature set to 20 C and the current supply temp is 22 C, then the controller has already entered the "Flow reduction" mode (visible also as the 3rd icon on the top left in the first overview page). You can see that your normal mode sets flow at 60%, but the controller has already reduced to 55% and possibly will fully stop at this point. This is what the controller does, that's all.
> 
> What I meant by does nothing is that ECO mode can be toggled in HA, but that does not change ECO status on the device (ECO button green when On and gray when Off), that I can see in WebApp, and vice versa, I can change ECO mode status on WebApp, but status in HA does not change.

Thanks for the clarifications. It definetly works for C6/C6M controllers. Once Add support for C8 controller type https://github.com/lnagel/hass-komfovent/pull/66 is merged, released and you find that it still does not work in the next version, then please open a ticket here in GitHub.

EvaldasD21, 2025-04-08:
> Thanks for the clarifications. It definetly works for C6/C6M controllers. Once Add support for C8 controller type [#66](https://github.com/lnagel/hass-komfovent/pull/66) is merged, released and you [...]

Thanks for your help.
I just noticed, that switching on/off ECO mode works, but response time (back and forth) is very long, ~1 minute.
Looking forward for update for C8 controller support.

lnagel, 2025-04-08:
> For operation mode, I do not see individual entities for different modes, there is only "climate.komfovent" entity which has drop down menu to select the mode, so no possibility to make custom card with mode buttons if I understand correctly.

The integration provides `select.komfovent_operation_mode` for indicating the mode. If you want to indicate current mode as button status, then you should be able to use some templating for each button to check the operation mode against a fixed value.

Setting a new value can be done via the `set_operation_mode` service call, which is documented in the README.

lnagel, 2025-04-08:
Released https://github.com/lnagel/hass-komfovent/releases/tag/v0.6.0

closed 2025-04-08 by lnagel